AUV-IITK/Varun-Software #26: (infrastructure) add udev rules

ShikherVerma, 2016-06-12:
add udev rules specific to hardware products so that we can access them on a fixed `tty`. like imu could be accessed at `/dev/navstik`. Also add it to setup script so it is added in all systems. Remove hardcoded path to hardware like imu code currently has a `PATH` file which has the path to imu which needs to be removed. .

ShikherVerma, 2016-06-16:
Also create a wiki page for adding a udev rule

pkhrag, 2016-07-06:
udev rules for imu has been added #78

ShikherVerma, 2016-07-06:
udev for arduino is left

pkhrag, 2016-07-07:
udev rule for arduino added in odroid. @ShikherVerma add the same to the setup script and make pull request.

ShikherVerma, 2016-07-07:
@pkhrag it is in odroid. we will add it the next time we boot it. Also we have to add in wiki the proceedure to get those udev rules. maybe a link to what you followed

ShikherVerma, 2016-07-09:
@pkhrag close this issue if this is done

ShikherVerma, 2016-08-13:
closed by #134

closed 2016-08-13 by ShikherVerma