AUV-IITK/Varun-Software #29: (task_handers_layer) Auto calibration image processing nodes.

ShikherVerma, 2016-06-12:
All image processing nodes should have a dynamic reconfigure button. on pressing it. image processing node calibrate its constants depending on the image from camera. Basically we tell the code that yes there is a specific arrangement in front of it. Now it needs to fix its constants such that it get starts getting positive results for that data.

siddsax, 2016-06-13:
Will do it for though for the time being you hard-code the parameters for a single colored ball. Need to make the code much more modular which I discovered when trying out the previous time, so it could be time consuming.

siddsax, 2016-06-22:
Final Result
6 while loops are taking a lot of time (more than 30 minutes) hence it is of not much use. We need to hard-code the range of different balls

closed 2016-06-22 by ShikherVerma

shibhansh, 2016-06-22:
siddharth send me the code that you have written. I'll try to optimize it.