AUV-IITK/Varun-Software #80: Documentation of Arduino code

shibhansh, 2016-07-06:
It should be in two levels one should be like an abstract and the other one should have detailed explanation for each function. It will help new members to contribute.

ShikherVerma, 2016-07-06:
@siddsax @soumye you guys are also going to do some part of documentation. You should also read the above comment.

siddsax, 2016-07-06:
I will write something for website so tell beforehand what you expect from it. I.E in the article(detailed one)

ShikherVerma, 2016-07-06:
@shibhansh ^

siddsax, 2016-07-06:
Read the comment from @shibhansh, though please make it more clear, i.e how much maths behind the methods (or just links to the documentation of the functions for the same), what you want to mean by how to use the algorithm. What should be the ideal length so that there can be an exact format for all documentations

ShikherVerma, 2016-07-06:
The doc should explain what the overall algorithm does and how. Not the explanation of how functions do what they do but just what the functions are. Like you could say that you do contrast stretching -> thresholding -> cannyEdge detection -> bla bla bla

shibhansh, 2016-07-06:
@siddsax for the code part it should do as @ShikherVerma said.
I will suggest that we should give the links wherever there is some algorithm. Also for IP nodes we can include the results in the documentation.

closed 2016-11-29 by ShikherVerma